Lifelong-Robot-Learning/LIBERO #11: Flipped demonstration images

minoring, 2023-11-05:
Hello, thank you for your excellent work!

I have a concern that might be incorrect, but it appears that the images in the demonstration dataset, both agentview_rgb and eye_in_hand_rgb, seem to be flipped upside down. I noticed this when comparing the first image from env.reset() with a second image from the demonstration data.

[image]
[image]

For reference, this is the code I used to visualize the demonstration data:
```python
import h5py
import cv2
import numpy as np
import argparse

def main():
    # Open the H5 file
    parser = argparse.ArgumentParser()
    parser.add_argument(
        "--dataset",
        type=str,
        help="path to hdf5 dataset",
    )
    args = parser.parse_args()

    with h5py.File(args.dataset, "r") as f:

        agentview_images = f["data/demo_0/obs/agentview_rgb"]
     
        # Display each image in the sequence
        for img in agentview_images:
            # Make video upside down.
            # img = np.flipud(img)
            cv2.imshow("AgentView Image", cv2.cvtColor(img, cv2.COLOR_RGB2BGR))
            cv2.waitKey(100)

        cv2.destroyAllWindows()

if __name__ == '__main__':
    main()
```

zhuyifengzju, 2023-11-11:
Hi, in the codebase, we use the gl image rendering by default (see notebook example: https://github.com/Lifelong-Robot-Learning/LIBERO/blob/master/notebooks/quick_walkthrough.ipynb). Can you double-check if you are using the same image rendering method between both reading from `env.reset()` and from demonstration data?

minoring, 2023-11-20:
Thank you for your response! I'm curious about whether `grid_image.numpy()[::-1]` flips the image vertically in your shared notebook. Could you please provide some clarification on this? Thanks!

zhuyifengzju, 2023-11-20:
Because we are using gl version as default, the image is upside down using the jupyter notebook visualization. It doesn't matter which function to use to visualize images, and you will need to flip images based on if a function assumes gl convention or the cv convention. The important thing is to keep the format of images consistent.

minoring, 2023-12-16:
Thank you for your response. I'm closing this issue as everything has been clarified now!

closed 2023-12-16 by minoring

Kashu7100, 2024-09-03:
I am wondering if using `grid_image.numpy()[::-1]` will flip the axis (not only the image) - instead of rotating 180 degrees.