Lifelong-Robot-Learning/LIBERO #14: Image rendering issues

jesbu1, 2024-01-25:
Hi, 

Sometimes when I'm using LIBERO environments (doesn't matter which suite), the images I get from the environment are look like white noise partially through the rollout, or they flip upside down, or even sometimes the `agentview_image` returns the in hand image (and I'm positive the key is correctly passed). Most of the times, most images are normal.

Some examples attached:

https://github.com/Lifelong-Robot-Learning/LIBERO/assets/12724137/7d8ea803-6518-4bb1-9fc3-064fb5627cb4

https://github.com/Lifelong-Robot-Learning/LIBERO/assets/12724137/e7883717-165c-4899-916a-420ddacee7cf

This could possibly be a problem with my custom code somewhere, so just curious if you ever encountered something like this?

jesbu1, 2024-01-25:
Another really strange rendering issue example attached, seems to happen at random 

https://github.com/Lifelong-Robot-Learning/LIBERO/assets/12724137/092f00f3-b778-4191-ae3e-f918f0cdc26c

closed 2024-01-29 by jesbu1

mw9385, 2025-12-05:
@xiaoen0  Hi, i got the same issue. Did you solve it?