Lifelong-Robot-Learning/LIBERO #18: the version numbers of robosuite and robomimic?

BrightMoonStar, 2024-03-30:
Could you please provide the version numbers of robosuite and robomimic?  I learned that robosuite has undergone major changes in versions 1.3 and 1.4, which may cause data sets to mismatch.  
Thanks a lot!
https://github.com/UT-Austin-RPL/VIOLA#dataset-and-trained-checkpoints  
 https://github.com/ARISE-Initiative/robosuite-task-zoo/issues/5#issuecomment-1965706715
[image]
[image]

Cranial-XIX, 2024-03-30:
LIBERO is compatible with v1.4.0 so no worries :)

BrightMoonStar, 2024-03-31:
Thanks for your reply !

closed 2024-03-31 by BrightMoonStar