Lifelong-Robot-Learning/LIBERO #19: TypeError: h5py objects cannot be pickled

BrightMoonStar, 2024-04-11:
Hi~
When I set num_worker to a number greater than 0, the following error will appear. How to solve this problem? I know that num_worker=0 can run, but I want to further improve the running speed. Thank you very much！

```
[info] start training on task 0
Error executing job with overrides: []
Traceback (most recent call last):
  File "/home/a/Videos/bin/boot/LIBERO/libero/lifelong/main.py", line 223, in main
    s_fwd, l_fwd = algo.learn_one_task(
  File "/home/a/Videos/bin/boot/LIBERO/libero/lifelong/algos/base.py", line 175, in learn_one_task
    for (idx, data) in enumerate(train_dataloader):
  File "/home/a/anaconda3/envs/LIBERO/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 439, in __iter__
    return self._get_iterator()
  File "/home/a/anaconda3/envs/LIBERO/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 387, in _get_iterator
    return _MultiProcessingDataLoaderIter(self)
  File "/home/a/anaconda3/envs/LIBERO/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 1040, in __init__
    w.start()
  File "/home/a/anaconda3/envs/LIBERO/lib/python3.8/multiprocessing/process.py", line 121, in start
    self._popen = self._Popen(self)
  File "/home/a/anaconda3/envs/LIBERO/lib/python3.8/multiprocessing/context.py", line 224, in _Popen
    return _default_context.get_context().Process._Popen(process_obj)
  File "/home/a/anaconda3/envs/LIBERO/lib/python3.8/multiprocessing/context.py", line 284, in _Popen
    return Popen(process_obj)
  File "/home/a/anaconda3/envs/LIBERO/lib/python3.8/multiprocessing/popen_spawn_posix.py", line 32, in __init__
    super().__init__(process_obj)
  File "/home/a/anaconda3/envs/LIBERO/lib/python3.8/multiprocessing/popen_fork.py", line 19, in __init__
    self._launch(process_obj)
  File "/home/a/anaconda3/envs/LIBERO/lib/python3.8/multiprocessing/popen_spawn_posix.py", line 47, in _launch
    reduction.dump(process_obj, fp)
  File "/home/a/anaconda3/envs/LIBERO/lib/python3.8/multiprocessing/reduction.py", line 60, in dump
    ForkingPickler(file, protocol).dump(obj)
  File "/home/a/anaconda3/envs/LIBERO/lib/python3.8/site-packages/h5py/_hl/base.py", line 368, in __getnewargs__
    raise TypeError("h5py objects cannot be pickled")
TypeError: h5py objects cannot be pickled

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.
```

Cranial-XIX, 2024-04-13:
Hi, could you please provide the full command to reproduce this error?

BrightMoonStar, 2024-04-13:
The following is the complete information when set num_worker>0 in .yaml file , so I have to set num_worker=0 to make it run which also causes training efficiency  lower as a side effect.

```
(LIBERO) lyf@Precision-7920-Tower:~/Videos/bin/boot/LIBERO/libero$ export CUDA_VISIBLE_DEVICES=0 && export MUJOCO_EGL_DEVICE_ID=0 && python lifelong/main.py seed=0 benchmark_name=libero_object policy=bc_rnn_policy lifelong=base
{ 'bddl_folder': None,
  'benchmark_name': 'libero_object',
  'data': { 'action_scale': 1.0,
            'affine_translate': 4,
            'data_modality': ['image', 'proprio'],
            'frame_stack': 1,
            'img_h': 128,
            'img_w': 128,
            'max_word_len': 25,
            'num_kp': 64,
            'obs': { 'modality': { 'depth': [],
                                   'low_dim': [ 'gripper_states',
                                                'joint_states'],
                                   'rgb': [ 'agentview_rgb',
                                            'eye_in_hand_rgb']}},
            'obs_key_mapping': { 'agentview_rgb': 'agentview_image',
                                 'eye_in_hand_rgb': 'robot0_eye_in_hand_image',
                                 'gripper_states': 'robot0_gripper_qpos',
                                 'joint_states': 'robot0_joint_pos'},
            'seq_len': 10,
            'shuffle_task': False,
            'state_dim': None,
            'task_group_size': 1,
            'task_order_index': 0,
            'train_dataset_ratio': 0.8,
            'use_ee': False,
            'use_eye_in_hand': True,
            'use_gripper': True,
            'use_joint': True},
  'device': 'cuda',
  'eval': { 'batch_size': 48,
            'eval': True,
            'eval_every': 5,
            'load_path': '',
            'max_steps': 600,
            'n_eval': 20,
            'num_procs': 20,
            'num_workers': 0,
            'save_sim_states': False,
            'use_mp': True},
  'folder': None,
  'init_states_folder': None,
  'lifelong': {'algo': 'Sequential'},
  'load_previous_model': False,
  'policy': { 'color_aug': { 'network': 'BatchWiseImgColorJitterAug',
                             'network_kwargs': { 'brightness': 0.3,
                                                 'contrast': 0.3,
                                                 'epsilon': 0.1,
                                                 'hue': 0.3,
                                                 'input_shape': None,
                                                 'saturation': 0.3}},
              'image_embed_size': 64,
              'image_encoder': { 'network': 'ResnetEncoder',
                                 'network_kwargs': { 'freeze': False,
                                                     'language_fusion': 'film',
                                                     'no_stride': False,
                                                     'pretrained': False,
                                                     'remove_layer_num': 4}},
              'language_encoder': { 'network': 'MLPEncoder',
                                    'network_kwargs': { 'hidden_size': 128,
                                                        'input_size': 768,
                                                        'num_layers': 1,
                                                        'output_size': 128}},
              'policy_head': { 'loss_kwargs': {'loss_coef': 1.0},
                               'network': 'GMMHead',
                               'network_kwargs': { 'activation': 'softplus',
[... 125 lines cut ...]
  File "/home/lyf/anaconda3/envs/LIBERO/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 387, in _get_iterator
    return _MultiProcessingDataLoaderIter(self)
  File "/home/lyf/anaconda3/envs/LIBERO/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 1040, in __init__
    w.start()
  File "/home/lyf/anaconda3/envs/LIBERO/lib/python3.8/multiprocessing/process.py", line 121, in start
    self._popen = self._Popen(self)
  File "/home/lyf/anaconda3/envs/LIBERO/lib/python3.8/multiprocessing/context.py", line 224, in _Popen
    return _default_context.get_context().Process._Popen(process_obj)
  File "/home/lyf/anaconda3/envs/LIBERO/lib/python3.8/multiprocessing/context.py", line 284, in _Popen
    return Popen(process_obj)
  File "/home/lyf/anaconda3/envs/LIBERO/lib/python3.8/multiprocessing/popen_spawn_posix.py", line 32, in __init__
    super().__init__(process_obj)
  File "/home/lyf/anaconda3/envs/LIBERO/lib/python3.8/multiprocessing/popen_fork.py", line 19, in __init__
    self._launch(process_obj)
  File "/home/lyf/anaconda3/envs/LIBERO/lib/python3.8/multiprocessing/popen_spawn_posix.py", line 47, in _launch
    reduction.dump(process_obj, fp)
  File "/home/lyf/anaconda3/envs/LIBERO/lib/python3.8/multiprocessing/reduction.py", line 60, in dump
    ForkingPickler(file, protocol).dump(obj)
  File "/home/lyf/anaconda3/envs/LIBERO/lib/python3.8/site-packages/h5py/_hl/base.py", line 368, in __getnewargs__
    raise TypeError("h5py objects cannot be pickled")
TypeError: h5py objects cannot be pickled

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.

```

raymondyu5, 2024-04-22:
Hi, I am also receiving this problem when I run:
```
export CUDA_VISIBLE_DEVICES=0 && export MUJOCO_EGL_DEVICE_ID=0 && python lifelong/main.py seed=0 benchmark_name=libero_90 policy=bc_rnn_policy lifelong=base
```
I receive this:
```
=================== Lifelong Benchmark Information  ===================
 Name: libero_90
 # Tasks: 90
    - Task 1:
        close the top drawer of the cabinet
    - Task 2:
        close the top drawer of the cabinet and put the black bowl on top of it
    - Task 3:
        put the black bowl in the top drawer of the cabinet
    - Task 4:
        put the butter at the back in the top drawer of the cabinet and close it
    - Task 5:
        put the butter at the front in the top drawer of the cabinet and close it
    - Task 6:
        put the chocolate pudding in the top drawer of the cabinet and close it
    - Task 7:
        open the bottom drawer of the cabinet
    - Task 8:
        open the top drawer of the cabinet
    - Task 9:
        open the top drawer of the cabinet and put the bowl in it
    - Task 10:
        put the black bowl on the plate
    - Task 11:
        put the black bowl on top of the cabinet
    - Task 12:
        open the top drawer of the cabinet
    - Task 13:
        put the black bowl at the back on the plate
    - Task 14:
        put the black bowl at the front on the plate
    - Task 15:
        put the middle black bowl on the plate
    - Task 16:
        put the middle black bowl on top of the cabinet
    - Task 17:
        stack the black bowl at the front on the black bowl in the middle
    - Task 18:
        stack the middle black bowl on the back black bowl
    - Task 19:
        put the frying pan on the stove
    - Task 20:
        put the moka pot on the stove
    - Task 21:
        turn on the stove
    - Task 22:
        turn on the stove and put the frying pan on it
    - Task 23:
        close the bottom drawer of the cabinet
    - Task 24:
        close the bottom drawer of the cabinet and open the top drawer
    - Task 25:
        put the black bowl in the bottom drawer of the cabinet
    - Task 26:
        put the black bowl on top of the cabinet
    - Task 27:
        put the wine bottle in the bottom drawer of the cabinet
    - Task 28:
        put the wine bottle on the wine rack
    - Task 29:
        close the top drawer of the cabinet
    - Task 30:
        put the black bowl in the top drawer of the cabinet
    - Task 31:
[... 142 lines cut ...]
  File "/home/raymond/anaconda3/envs/libero/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 314, in _get_iterator
    return _MultiProcessingDataLoaderIter(self)
  File "/home/raymond/anaconda3/envs/libero/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 927, in __init__
    w.start()
  File "/home/raymond/anaconda3/envs/libero/lib/python3.8/multiprocessing/process.py", line 121, in start
    self._popen = self._Popen(self)
  File "/home/raymond/anaconda3/envs/libero/lib/python3.8/multiprocessing/context.py", line 224, in _Popen
    return _default_context.get_context().Process._Popen(process_obj)
  File "/home/raymond/anaconda3/envs/libero/lib/python3.8/multiprocessing/context.py", line 284, in _Popen
    return Popen(process_obj)
  File "/home/raymond/anaconda3/envs/libero/lib/python3.8/multiprocessing/popen_spawn_posix.py", line 32, in __init__
    super().__init__(process_obj)
  File "/home/raymond/anaconda3/envs/libero/lib/python3.8/multiprocessing/popen_fork.py", line 19, in __init__
    self._launch(process_obj)
  File "/home/raymond/anaconda3/envs/libero/lib/python3.8/multiprocessing/popen_spawn_posix.py", line 47, in _launch
    reduction.dump(process_obj, fp)
  File "/home/raymond/anaconda3/envs/libero/lib/python3.8/multiprocessing/reduction.py", line 60, in dump
    ForkingPickler(file, protocol).dump(obj)
  File "/home/raymond/anaconda3/envs/libero/lib/python3.8/site-packages/h5py/_hl/base.py", line 370, in __getnewargs__
    raise TypeError("h5py objects cannot be pickled")
TypeError: h5py objects cannot be pickled

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.
```
Please let me know, thanks!

bit0123, 2024-05-11:
Hi @Cranial-XIX 
Thanks for the interesting paper and releasing the code.
I am facing the same issue ("TypeError: h5py objects cannot be pickled") while running with num_worker=4. Although the code works with num_worker=0 with some additional transformation of image input in the policy file, The loss becomes negative after the first epoch for different algorithm. Consequently, the accuracy is 0. 
Could you please comment on that.
Thanks

modanesh, 2024-05-22:
@Cranial-XIX Any solutions?

modanesh, 2024-05-22:
Tried all configurations of `algo`, `policy`, and `benchmark` and still got the same error.

raymondyu5, 2024-05-22:
Try adding these two lines to venv.py. This took me like two days to figure out, so I hope this helps!
if multiprocessing.get_start_method(allow_none=True) != "spawn":  
    multiprocessing.set_start_method("spawn", force=True)

modanesh, 2024-05-22:
@raymondyu5 Thanks for the response. Where should I exactly add these lines?

Btw, from my understanding, `venv.py` file is not used when running the training (`libero/lifelong/main.py`) which is what I want to do.

bit0123, 2024-05-24:
Hi @BrightMoonStar and @raymondyu5 
Could you please comment on the following observation. 
"The code works with num_worker=0, however the loss becomes negative after the first epoch for different algorithm. Consequently, succes and the AoC is 0."
Thanks for your response.

wangair, 2024-05-27:
Hi!
I met the same problem.I set num_worker=0, but it didn't work for me. When i set:
`num_worker=0`
I received this:
> ValueError: persistent_workers option needs num_workers > 0
How can i solve this error?Thank you very much.

BrightMoonStar, 2024-06-24:
> 你好@BrightMoonStar，@raymondyu5和@Cranial-XIX 您能否使用 ER 算法对以下观察进行评论。 “代码在 num_worker=0 的情况下有效，但是对于不同的算法，在第一个 epoch 之后损失变为负值。因此，成功且 AoC 为 0。” 感谢您的回复。

I meet the same problem

BrightMoonStar, 2024-06-24:
> 你好@BrightMoonStar，@raymondyu5和@Cranial-XIX 您能否使用 ER 算法对以下观察进行评论。 “代码在 num_worker=0 的情况下有效，但是对于不同的算法，在第一个 epoch 之后损失变为负值。因此，成功且 AoC 为 0。” 感谢您的回复。

Have you solved it？Thanks for your reply!

pengzhi1998, 2024-06-25:
Dear all,

I faced the same issue. And I found here a solution: https://github.com/pytorch/pytorch/issues/11929 and https://discuss.pytorch.org/t/hdf5-multi-threaded-alternative/6189/9. They both mentioned we need to open the hdf5 file in `__getitem__` method but not `__init__` of the custom `Dataset` if using multi-processing dataloader. Thus, I modified directly on [SequenceDataset from RoboMimic](https://github.com/ARISE-Initiative/robomimic/blob/v0.3.0/robomimic/utils/dataset.py#L18). However, the problem of "TypeError: h5py objects cannot be pickled" still exists. Have you solved this issue?

May I have your suggestions on this? @Cranial-XIX @zhuyifengzju 

Thank you for your attention on this matter and for your precious time!!

Best regards,
Pengzhi

pengzhi1998, 2024-06-27:
> Hi! I met the same problem.I set num_worker=0, but it didn't work for me. How can i solve this error?Thank you very much.When i set: `num_worker=0` I received this:
> 
> > ValueError: persistent_workers option needs num_workers > 0

Hi, I commented two lines when creating the dataloader in base.py:
```
        train_dataloader = DataLoader(
            dataset,
            batch_size=self.cfg.train.batch_size,
            # num_workers=self.cfg.train.num_workers,
            sampler=RandomSampler(dataset),
            # persistent_workers=False,
        )
```

Cranial-XIX, 2024-06-29:
Hi, May I ask if you are using a Linux or Windows machine to run it?

Cranial-XIX, 2024-06-29:
Can you comment the

```
if multiprocessing.get_start_method(allow_none=True) != "spawn":
    multiprocessing.set_start_method("spawn", force=True) 
```

From 270-271 in `libero/lifelong/main.py` and try again?

pengzhi1998, 2024-06-30:
Thank you so much for your reply!

Your solution worked! However, if I comment out the two lines, the issue reappears (https://github.com/Lifelong-Robot-Learning/LIBERO/issues/3).

I would first try downgrading the Cuda/Driver version. Besides this method, may I have your suggestions on this?

Thank you so much again. Look forward to your reply!

Junxix, 2024-10-17:
> Can you comment the
> 
> ```
> if multiprocessing.get_start_method(allow_none=True) != "spawn":
>     multiprocessing.set_start_method("spawn", force=True) 
> ```
> 
> From 270-271 in `libero/lifelong/main.py` and try again? Try keeping everything else the same as in the HEAD first. It works fine on my side.

I still get the same problem after adding the two lines. Do you know how to solve it?

Kami-code, 2025-04-10:
I have encountered the same problem on my 3.10 environment.

For my 3.8 environment, I don't have to do any modification and it works. 
For the 3.10 environment, I commented the 

`if multiprocessing.get_start_method(allow_none=True) != "spawn":
    multiprocessing.set_start_method("spawn", force=True) `

and it starts training.

zsc0731, 2026-03-26:
it's really helpful:(changing the base.py)
train_dataloader = DataLoader(
            dataset,
            batch_size=self.cfg.train.batch_size,
            num_workers=0,                # 0
            sampler=RandomSampler(dataset),
            persistent_workers=False      # False
        )