Lifelong-Robot-Learning/LIBERO #21: Negative Loss, Zero Success and Zero AoC

bit0123, 2024-05-17:
Hi,
Thanks for the interesting paper and releasing the code.
I am facing the same issue ("TypeError: h5py objects cannot be pickled") while running with num_worker=4. However, the code works when i set num_worker=0 with some additional transformation of input image in the policy file. 

The loss becomes negative after the first epoch for different algorithm with 0 success. Consequently, the AoC is 0.
Could you please comment on that.
Thanks

wangair, 2024-06-14:
> Hi, Thanks for the interesting paper and releasing the code. I am facing the same issue ("TypeError: h5py objects cannot be pickled") while running with num_worker=4. However, the code works when i [...]

Hi!Have you solved this?

BrightMoonStar, 2024-06-23:
I encountered the same problem. All the evaluation indicators were 0, including AOC, Success Rate ...

pengzhi1998, 2024-06-25:
Hi, I'm facing the same problem: https://github.com/Lifelong-Robot-Learning/LIBERO/issues/19#issuecomment-2188322104. Any suggestions?

BrightMoonStar, 2024-06-25:
What confuses me is that the following two projects based on LIBERO also have success rate = 0 and Aoc = 0. No matter how we train, the result is 0. Is it because the details of our configuration are incorrect? I would like to ask for your help. Thank you very much!  @Cranial-XIX 
https://github.com/UT-Austin-RPL/Lotus
https://github.com/FrankZheng2022/PRISE

pengzhi1998, 2024-06-28:
Oh, I found after 20 epochs of training, the success rate would become larger than 0. And after 50 epochs of training, the average success rate is around 0.7. (for `libero_object` tasks)

BrightMoonStar, 2024-06-30:
> 经过 20 个时期的训练后，成功率将大于 0。经过 50 个时期的训练后，平均成功率约为 0.7。（对于`libero_object`任务）

Have you ever tried libero-90? For libero-90, it seems the problem still exists for me. Thank you very much!

Cranial-XIX, 2024-06-30:
The success rate on libero-90 is low by design (since there are 90 tasks), and in general, you might need over 20 epochs (like 25 - 35) to reach the peak result.