Lifelong-Robot-Learning/LIBERO #24: Calling env.reset() multiple times causing kernel crash

JummerCloth, 2024-07-06:
Hello authors! Thanks for this wonderful contribution to the robot learning community. I have been trying to use this simulator for some evaluation on a model I trained, but the system seems to bug out every time after a while. Later investigation shows that after calling env.reset() around 20 times, it will never succeed, resulting in infinite ValueError. I have reproduced a similar issue with a small modification of the getting started code provided on your README file: 
[dummy.py.zip](https://github.com/user-attachments/files/16114544/dummy.py.zip)

With this simple script, the terminal keeps outputting env reset after around 20 iterations. Could you please look into this issue? Much thanks! I am using the libero env provided, with the following machine specifications:
Architecture:                    x86_64
CPU op-mode(s):                  32-bit, 64-bit
Byte Order:                      Little Endian
Address sizes:                   43 bits physical, 48 bits virtual
CPU(s):                          96
On-line CPU(s) list:             0-95
Thread(s) per core:              2
Core(s) per socket:              24
Socket(s):                       2
NUMA node(s):                    2
Vendor ID:                       AuthenticAMD
CPU family:                      23
Model:                           49
Model name:                      AMD EPYC 7402 24-Core Processor
Stepping:                        0
Frequency boost:                 enabled
CPU MHz:                         1497.918
CPU max MHz:                     2800.0000
CPU min MHz:                     1500.0000
BogoMIPS:                        5599.99
Virtualization:                  AMD-V
L1d cache:                       1.5 MiB
L1i cache:                       1.5 MiB
L2 cache:                        24 MiB
L3 cache:                        256 MiB
NUMA node0 CPU(s):               0-23,48-71
NUMA node1 CPU(s):               24-47,72-95

JummerCloth, 2024-07-22:
I resolved the issue by upgrading the robosuite: `pip install --force-reinstall robosuite`, and downgrading numpy to a 1.x version.

closed 2024-07-22 by JummerCloth

pengzhi1998, 2024-07-29:
Hi, previously I didn't have this issue. But this week it suddenly happened to me as well which is very wired. 
Thank you for your solution! However, may I ask which Numpy version you are using specifically?