Lifelong-Robot-Learning/LIBERO #26: Understanding the state and action spaces of demonstration data

Mr-vedant-gupta, 2024-07-31:
Hi!

[image]

igeorgiev-bdai, 2024-08-16:
Hi, did you figure out what the action space is in the end? Is it really change in EE pose?

Mr-vedant-gupta, 2024-08-18:
Hi @igeorgiev-bdai 
I'm still waiting for a response on this post, but believe the action space is change in EE pose. 
See [here](https://robosuite.ai/docs/modules/controllers.htmlposehttps://robosuite.ai/docs/modules/controllers.html): When using any position-based control (OSC, IK, or JOINT_POSITION controllers), input actions are, by default, interpreted as delta values from the current state. 
Based on some digging through the code I remember the action space being OSC_POSE

zhuyifengzju, 2024-08-18:
Hi,
The action space is the OSC_POSE. I am not sure what you mean by the action space being Markovian --- Typically we can if a policy is a markovian or the state space is markovian. Can you elaborate on this? 

Regarding the observation keys:
- `XXX_rgb`: The RGB image of camera named under `XXX`
- ee_ori: axis angles of end effectors
- ee_pos: locations of end effectors
- ee_states: the concatenation of ee_pos and ee_ori
- gripper_states: the joint readings of the gripper
- joint_states: joint angle readings of the robot arm

The `states` here refer to the mujoco simulation states, which is useful if you want to directly reset the environment to a specific timeframe. The dimension 47 means that for this task, the number of joints of the entire simulation is 47. For this one, please refer to the mujoco  documentation for more details.

closed 2024-09-07 by Cranial-XIX

YEOMJINSEOP, 2024-10-25:
@zhuyifengzju  Thanks for the detailed responses in previous comments. 

I have a question about **robot_states with shape (T, 9).** 
Could you clarify what  9 dimensions represents within this tensor?

Thanks again for the insightful work on this project.

zhuyifengzju, 2024-10-25:
do you mean the `joint_states`? If that's the case, I believe it's 7 (arm joint states) + 2 (gripper states), but since the gripper joint control is mirrored, the action space is only 1 for the gripper.

YEOMJINSEOP, 2024-10-25:
> responses in previous comments.

Thanks for reply @zhuyifengzju.

Referring to the image from the issue's initial comment, 
I’m trying to clarify the difference between:

- `robot_states` with shape (329, 9), which is the third line from the bottom.
- several states in `obs` including ee_states, gripper_states, joint states.
[image]

Based on your previous response, could I interpret `robot_states` with shape (329, 9) as a combination of **joint states** (329, 7) and **gripper states** (329, 2)?

I apologize if my question seems basic, as I am relatively new to robotics research.

Thanks in advance for your clarification!

zhuyifengzju, 2024-10-25:
Yes, I believe it's correct. It would be safe to double check yourself again (looking at the numericals)

GuHuangAI, 2024-12-17:
> Hi @igeorgiev-bdai I'm still waiting for a response on this post, but believe the action space is change in EE pose. See [here](https://robosuite.ai/docs/modules/controllers.html): "When using any position-based control (OSC, IK, or JOINT_POSITION controllers), input actions are, by default, interpreted as delta values from the current state" Based on some digging through the code I remember the action space being OSC_POSE
> 
> @Cranial-XIX Apologies for the tag but any info on my question would be very much appreciated as I have some research that depends on this!

Hello, I found that the difference of ee_state of two neighbouring frames was not equal to the corresponding action. Could you explain this?

GuHuangAI, 2024-12-17:
> Yes, I believe it's correct. It would be safe to double check yourself again (looking at the numericals)

Hello, what dose the action mean? I found that the difference of ee_state of two neighbouring frames was not equal to the corresponding action.

Lucky-Light-Sun, 2025-04-24:
> Yes, I believe it's correct. It would be safe to double check yourself again (looking at the numericals)

however, i checked this, maybe the `robot states` is no the concat of `gripper_states` and `joint_states`.

[image]

AnnabellaMacaluso, 2025-05-15:
Hi, is there any update on the specifics of what action here represents? It doesn't seem to represent end effector pose because the outputs are not achievable in real world. If it's delta pose, it's not in meters which was mentioned in prior issues. Urgently need help on this, thank you.