Lifelong-Robot-Learning/LIBERO #27: Question for creation of libero/libero/init_files

YY-GX, 2024-08-01:
Thank you for sharing this wonderful benchmark! 

However, I encountered some issues when creating my own dataset: I could not find any scripts helping create the `libero/libero/init_files` for me after I obtained the .hdf5 demos. Could you please show me how to generate these `.init` & `.pruned_init` files? Thank you!!

Cranial-XIX, 2024-08-01:
The init_files are created for controlling the randomness during evaluation, it stores the init states we randomly generated, so everyone has a fair comparison on this benchmark. If you create your customized dataset, you can just randomly initialize the environments 50 times and store the initial state, then during evaluation, you set it to those init states.

YY-GX, 2024-08-02:
Thank you for the quick reply!! 

So are those `.pruned_init` files just numpy arrays? What's the difference between the `.init` and `.pruned_init`? Thank you!

Cranial-XIX, 2024-09-07:
.init contains 50 of them, .pruned_init is 20 out of the 50 for evaluation purpose if I remember correctly :)

closed 2024-09-07 by Cranial-XIX