Lifelong-Robot-Learning/LIBERO #29: Adding depth to demo datasets

Aryan9101, 2024-08-05:
Hello, the current demo datasets don't include any depth maps. How should I go about recreating the dataset so it has depth maps as an observation along with the rgb images? Thanks!

prinshul, 2024-10-01:
@Cranial-XIX Same question!

yasibot8, 2024-10-12:
Same question!

yasibot8, 2024-10-12:
> Hello, the current demo datasets don't include any depth maps. How should I go about recreating the dataset so it has depth maps as an observation along with the rgb images? Thanks!

Did you get?

zhuyifengzju, 2024-10-12:
Hi, robosuite documentation will be updated soon with more detailed description. For now, I am providing the most critical parts that are relevant:
https://github.com/ARISE-Initiative/robosuite/blob/74981fd347660205efb4f8c0005e28ec621f5304/robosuite/environments/robot_env.py#L140

convert to a metric depth map: https://github.com/ARISE-Initiative/robosuite/blob/74981fd347660205efb4f8c0005e28ec621f5304/robosuite/utils/camera_utils.py#L106

After camera_depth is enabled, you should be able to see a key in `obs` named `CAMERA_NAME_depth`.

AndyCao1125, 2024-10-20:
@zhuyifengzju 
Hi! I have a question regarding the depth output when setting `camera_depth=True` in the Libero environment (e.g., goal, 10). According to the [RoboSuite depth documentation](https://robosuite.ai/docs/modules/renderers.html?highlight=depth), this should return normalized depth values in the range of [0, 1]. However, the observed depth values are roughly between [0.98, 0.99], resulting in a near-white image unless normalized.

Could you clarify whether the depth images returned in the Libero environment are raw metric depth or normalized depth? If they are raw depth values, what are their units, and how can I obtain the minimum and maximum depth information that the depth camera can capture?

Thanks!

closed 2024-10-20 by zhuyifengzju

ZeyuPing, 2025-03-07:
@AndyCao1125 Hi! Could you please offer me some codes for reference? I still don't know how to replay it T_T