Lifelong-Robot-Learning/LIBERO #30: An issue about dataset.

Zhangwenyao1, 2024-08-14:
Thanks for your great work!
Are there any segmentation masks and depth in your dataset.

Cranial-XIX, 2024-08-14:
Thanks for your interest in our work, currently we do not have these in our datasets. But those are provided in Robosuite, so I believe if you re-rollout the dataset you can get the mask/depth from the simulator.

Zhangwenyao1, 2024-08-14:
> Thanks for your interest in our work, currently we do not have these in our datasets. But those are provided in Robosuite, so I believe if you re-rollout the dataset you can get the mask/depth from the simulator.

Thanks for your reply.
Can you tell me how to re-rollout the dataset？
Are there any tutorials ?

Cranial-XIX, 2024-08-14:
You can take a look at our README page. Basically, you choose a task, set the initial state, and then simulate the action, and then get the observation with depth/mask.

Zhangwenyao1, 2024-08-14:
> You can take a look at our README page. Basically, you choose a task, set the initial state, and then simulate the action, and then get the observation with depth/mask.

Thanks for your quick reply!
Best wishes!

closed 2024-09-07 by Cranial-XIX