Lifelong-Robot-Learning/LIBERO #38: Installation error

arthurnni, 2024-10-05:
Hello,
I follow the installation instruction below

conda create -n libero python=3.8.13
conda activate libero
git clone https://github.com/Lifelong-Robot-Learning/LIBERO.git
cd LIBERO
pip install -r requirements.txt

While running " pip install -r requirements.txt "  there  are  errors as below

--------------------------------------------------------------------------------------------------------------

    Created wheel for antlr4-python3-runtime: filename=antlr4_python3_runtime-4.9.3-py3-none-any.whl size=144555 sha256=d373f0a90c4e416d712cbade96eb0302316d954bb49f5cbef5f104d5f1bd5521
  Stored in directory: /home/aries14/.cache/pip/wheels/b1/a3/c2/6df046c09459b73cc9bb6c4401b0be6c47048baf9a1617c485
  Building wheel for egl_probe (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [356 lines of output]
      /home/aries14/miniconda3/envs/libero/lib/python3.8/site-packages/setuptools/_distutils/dist.py:261: UserWarning: Unknown distribution option: 'tests_require'
        warnings.warn(msg)
      running bdist_wheel
      running build
      running build_py
      creating build/lib.linux-x86_64-cpython-38/egl_probe
      copying egl_probe/get_available_devices.py -> build/lib.linux-x86_64-cpython-38/egl_probe
      copying egl_probe/__init__.py -> build/lib.linux-x86_64-cpython-38/egl_probe
      running egg_info
      writing egl_probe.egg-info/PKG-INFO
      writing dependency_links to egl_probe.egg-info/dependency_links.txt
      writing top-level names to egl_probe.egg-info/top_level.txt
      reading manifest file 'egl_probe.egg-info/SOURCES.txt'
      reading manifest template 'MANIFEST.in'
      adding license file 'LICENSE'
      writing manifest file 'egl_probe.egg-info/SOURCES.txt'
      /home/aries14/miniconda3/envs/libero/lib/python3.8/site-packages/setuptools/command/build_py.py:218: _Warning: Package 'egl_probe.glad' is absent from the `packages` configuration.
      !!
      
              ********************************************************************************
              ############################
              # Package would be ignored #
              ############################
              Python recognizes 'egl_probe.glad' as an importable package[^1],
              but it is absent from setuptools' `packages` configuration.
      
              This leads to an ambiguous overall configuration. If you want to distribute this
              package, please make sure that 'egl_probe.glad' is explicitly added
              to the `packages` configuration field.
      
              Alternatively, you can also rely on setuptools' discovery methods
              (for example by using `find_namespace_packages(...)`/`find_namespace:`
              instead of `find_packages(...)`/`find:`).
      
              You can read more about "package discovery" on setuptools documentation page:
      
              - https://setuptools.pypa.io/en/latest/userguide/package_discovery.html
      
              If you don't want 'egl_probe.glad' to be distributed and are
              already explicitly excluding 'egl_probe.glad' via
              `find_namespace_packages(...)/find_namespace` or `find_packages(...)/find`,
              you can try to use `exclude_package_data`, or `include-package-data=False` in
              combination with a more fine grained `package-data` configuration.
      
              You can read more about "package data files" on setuptools documentation page:
      
              - https://setuptools.pypa.io/en/latest/userguide/datafiles.html
      
      
[... 352 lines cut ...]
      The 'linux/input.h' and 'linux/input-event-codes.h' include files
      are missing. You will have to install the kernel header files in
      order to continue:
      
          dnf install kernel-headers-$(uname -r)
          apt-get install linux-headers-$(uname -r)
          emerge sys-kernel/linux-headers
          pacman -S kernel-headers
      
      In case they are installed in a non-standard location, you may use
      the '--evdev-headers' option to specify one or more colon-separated
      paths. For example:
      
          python setup.py \
            build \
            build_ecodes --evdev-headers path/input.h:path/input-event-codes.h \
            build_ext --include-dirs path/ \
            install
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for evdev
Successfully built easydict robomimic bddl future gym antlr4-python3-runtime promise pathtools
Failed to build egl_probe evdev
ERROR: ERROR: Failed to build installable wheels for some pyproject.toml based projects (egl_probe, evdev)

zhuyifengzju, 2024-10-20:
Hi, can you first double check if you can successfully install robosuite and run environments from robosuite. This error message is hard to parse, but I think overall the issue comes from egl. Depending on where you are install the code, this egl thing either needs to be fixed from the system level, or ignored (e.g. if you are using mac, the egl should not be installed because egl is supported for linux).

Norrrrrrr-lyn, 2025-04-18:
I've encountered the same error, did you fix it?

Tonghe-Zhang, 2025-06-19:
Install egl_probe from a correct branch by doing the following: 
```bash
# install cmake<4.0.0
pip install cmake==3.31.6
# download egl_prob from another branch: (ensure internet connection)
wget https://github.com/mhandb/egl_probe/archive/fix_windows_build.zip
# the install egl_prob from another branch:
pip install fix_windows_build.zip
```
Then re-install your LIBERO or robomimic again, it should be fine.

purentap, 2025-07-17:
I had the same error, tried installing robosuite seperately but the error persisted. what solved the issue was `sudo apt-get install build-essential` on ubuntu. hope it helps.