Lifelong-Robot-Learning/LIBERO #39: Negative Traing-loss

JokerESC, 2024-10-08:
Hello, Thanks for your great work!
Based on the bc-transformer-policy and multitask learning algorithm in the document, I conducted 50 epochs of training experiments on five data sets. The success rate is basically consistent with the results provided by the paper, but the training loss is negative. For example, the loss of libero-object converges to about -20. Why does this happen?

zhuyifengzju, 2024-10-20:
Hi, there is a common misunderstanding that the training loss has to be positive --- that's the case when the prediction is a categorical distribution for classification tasks. However, for the action space of libero, the loglikelihood is computed over actions, which range from [-1, 1]. See https://github.com/Lifelong-Robot-Learning/LIBERO/blob/f78abd68ee283de9f9be3c8f7e2a9ad60246e95c/libero/lifelong/models/policy_head.py#L100 .

JokerESC, 2024-10-21:
> Hi, there is a common misunderstanding that the training loss has to be positive --- that's the case when the prediction is a categorical distribution for classification tasks. However, for the action space of libero, the loglikelihood is computed over actions, which range from [-1, 1] for each dimension. See
> 
> https://github.com/Lifelong-Robot-Learning/LIBERO/blob/f78abd68ee283de9f9be3c8f7e2a9ad60246e95c/libero/lifelong/models/policy_head.py#L100

Thanks for your answer! If the input action is standardized, or the entire data set is standardized, and a positive loss is obtained, will it affect the training?

zhuyifengzju, 2024-10-21:
Can you elaborate what do you mean "standardized"? You can postprocess actions if you want, but make sure you map the postprocessed actions back to the original action space for the simulator to work properly.

JokerESC, 2024-10-21:
What I mean is that using some general methods of processing data, such as subtracting the mean value from the data set and dividing it by the standard deviation, and obtaining a positive loss during the training process, will it have a positive impact on the training?

zhuyifengzju, 2024-10-21:
It might be helpful if you only evaluate policies on the training task distributions --- but if you are considering lifelong learning setting, then you will not have the mean value of the datasets of unseen tasks a priori, therefore, might cause worse forward transfer. I think there are no restrictions for you to postprocess the datasets, as you mentioned. Just proceed with cautious that you do not leak the statistics of "unseen" task demonstrations when you train policies on training tasks.

JokerESC, 2024-10-21:
I understand, thanks again for your reply！

closed 2024-10-21 by JokerESC