Lifelong-Robot-Learning/LIBERO #43: Absolute control for LIBERO

kwanyoungpark, 2024-10-31:
I would like to use absolute EEF control for LIBERO.

I've tried with delta control (calculate the conversion from absoulte -> delta action by myself), but I was unable to find the right formula.
The question is :
1) How to use absolute EEF control for LIBERO environments?
2) How can we change the action from delta EEF control to absolute EEF control in the dataset?

My (failed) conversion formula from delta EEF control to absolute EEF control is as follows:
```
absol_action_pos = action[:3] * 0.05 + obs['ee_pos']
absol_action_rot = mat2axisangle(np.matmul(axisangle2mat(action[3:6] * 0.5), axisangle2mat(obs['ee_ori'])))
```

Where `obs` and `action` is in the same time step (i.e. `action` is an action corresponding to observation `obs` )

Yiyao-Ma, 2025-04-01:
Hi, have you solved it? I got the same problem.

farazsrahman, 2025-06-01:
I am also wondering the same thing. Where did you get 0.05 from? I was told the delta ee_pose data is unnormalized in meters, but when I plot the data it ranges from -1 to 1?

WwZzz, 2025-06-23:
> I am also wondering the same thing. Where did you get 0.05 from? I was told the delta ee_pose data is unnormalized in meters, but when I plot the data it ranges from -1 to 1. Is the data actually normalized?

The value of 0.05 is from the function `scale_action(self, action)` in the package robosuite.

joonsu0109gh, 2025-08-25:
Same problem

zwbx, 2025-09-17:
same problem