Lifelong-Robot-Learning/LIBERO #45: gripper position

xavihart, 2024-11-13:
I am trying to get the trajectory of the gripper in the pixel space for demo videos, but failed for a long time. 

I tried to use `ee_pos` and 'ee_rot' and camera info from `demo.attrs['model_file']`, but the rendered result is bad; it seems that there is something wrong with the provided camera matrix or the ee_pos. (It seems that ee_pos is the pose of the end-effector of the franka robot arm, not the gripper)

Do you know there is any code that can help me visualize the gripper trajectory on the pixel space? Thanks.

vineet2104, 2025-02-20:
Hi, were you able to figure this out?

xavihart, 2025-02-20:
yes, i think i made some mistakes when doing camera projection

yfpeng1234, 2025-03-18:
Hi Haotian @xavihart , could you share how to achieve this?