Lifelong-Robot-Learning/LIBERO #48: collect_demonstrations.py

wangair, 2024-11-19:
Hello, thank you for your excellent contribution！I want to collect some demonstrations by running “collect_demonstrations.py --controller OSC_POSE   --camera agentview   --num-demonstration 50  --rot-sensitivity 1.5 --bddl-file /home/gc/LIBERO/libero/libero/bddl_files/libero_goal/turn_on_the_stove.bddl --robots Panda”. But I met the below error.How can I solve it?

 Traceback (most recent call last):
  File "scripts/collect_demonstration.py", line 315, in <module>
    env.viewer.add_keypress_callback("any", device.on_press)
TypeError: add_keypress_callback() takes 2 positional arguments but 3 were given

GuHuangAI, 2024-12-24:
@wangair Hello, have you solved this problem?

GuHuangAI, 2024-12-24:
@wangair  This problem occurred when your robosuite's version >= 1.4.0. 
Finally, I solved this issue by commenting the line 315-317, instead, add "env.viewer.add_keypress_callback(device.on_press)"

YancieYang, 2025-03-17:
I change the code, but it triggers problem:
Traceback (most recent call last):
  File "scripts/collect_demonstration.py", line 320, in <module>
    env.viewer.add_keyup_callback(device.on_release)
AttributeError: 'OpenCVRenderer' object has no attribute 'add_keyup_callback'