Lifelong-Robot-Learning/LIBERO #49: ModuleNotFoundError: No module named 'robosuite.environments.manipulation.single_arm_env'

zavorotnyi, 2024-11-20:
The SingleArmEnv was removed from  [robosuite](https://github.com/ARISE-Initiative/robosuite) in transition from 1.4 to 1.5 see issue: https://github.com/ARISE-Initiative/robosuite/issues/560

ModuleNotFoundError while trying to import OffScreenRenderEnv from libero.libero.envs 

` 
File "robot/libero/libero_utils.py", line 10, in <module>
    from libero.libero.envs import OffScreenRenderEnv
  File "LIBERO/libero/libero/envs/__init__.py", line 1, in <module>
    from .bddl_base_domain import TASK_MAPPING
  File "LIBERO/libero/libero/envs/bddl_base_domain.py", line 6, in <module>
    from robosuite.environments.manipulation.single_arm_env import SingleArmEnv
ModuleNotFoundError: No module named 'robosuite.environments.manipulation.single_arm_env'
`

object814, 2024-12-09:
Thanks a lot, I encountered the same problem and by downgrading robosuite to 1.4.0 it worked.

zavorotnyi, 2025-04-30:
https://github.com/Lifelong-Robot-Learning/LIBERO/pull/51

closed 2025-04-30 by zavorotnyi