Lifelong-Robot-Learning/LIBERO #54: Does the action precede the observation?

GuoPingPan, 2024-12-24:
I found that when collecting data, observation is the result after env.step(action). Does it mean that every pair in the dataset is (a_t, o_{t+1})? @zhuyifengzju 

https://github.com/Lifelong-Robot-Learning/LIBERO/blob/f78abd68ee283de9f9be3c8f7e2a9ad60246e95c/scripts/create_dataset.py#L177
```
   for j, action in enumerate(actions):

            obs, reward, done, info = env.step(action)

            if j < num_actions - 1:
                # ensure that the actions deterministically lead to the same recorded states
                state_playback = env.sim.get_state().flatten()
                # assert(np.all(np.equal(states[j + 1], state_playback)))
                err = np.linalg.norm(states[j + 1] - state_playback)

                if err > 0.01:
                    print(
                        f"[warning] playback diverged by {err:.2f} for ep {ep} at step {j}"
                    )

            # Skip recording because the force sensor is not stable in
            # the beginning
            if j < cap_index:
                continue

            valid_index.append(j)

            if not args.no_proprio:
                if "robot0_gripper_qpos" in obs:
                    gripper_states.append(obs["robot0_gripper_qpos"])

                joint_states.append(obs["robot0_joint_pos"])

                ee_states.append(
                    np.hstack(
                        (
                            obs["robot0_eef_pos"],
                            T.quat2axisangle(obs["robot0_eef_quat"]),
                        )
                    )
                )

            robot_states.append(env.get_robot_state_vector(obs))

            if args.use_camera_obs:

                if args.use_depth:
                    agentview_depths.append(obs["agentview_depth"])
                    eye_in_hand_depths.append(obs["robot0_eye_in_hand_depth"])

                agentview_images.append(obs["agentview_image"])
                eye_in_hand_images.append(obs["robot0_eye_in_hand_image"])
            else:
                env.render()
```

MarSaKi, 2025-05-07:
@GuoPingPan @zhuyifengzju 
Hi, have you find the answer? Is the action precede the observation? 
thanks,

zhuyifengzju, 2025-05-07:
Hello,

Thanks for pointing this out. From the code, it definitely seems to be wrong in terms of implementation logic. It's been quite a while since we developed the whole repo, and I cannot find the original script for creating datasets for reference before cleaning up code logic. As for datasets, as long as they replay and result in success, the trajectories can be used for doing behavioral cloning. If you want to create your own datasets, please proceed with caution or flip the logic. 

Thanks,

toannguyen1904, 2025-10-03:
Hello @zhuyifengzju, @GuoPingPan, do we have an answer for this question?

Xj1567589354, 2026-01-18:
What's the current situation? Are there any problems with the official collect_demonstration.py and create_dataset.py files?