Lifelong-Robot-Learning/LIBERO #56: AttributeError: 'MjRenderContextOffscreen' object has no attribute 'con'

FuNz-0, 2025-01-04:
[image]
What's the problem

gray311, 2025-02-10:
same question

vineet2104, 2025-03-25:
Were you able to solve this?

arnavbalaji, 2025-03-27:
Any update on this?

CostaliyA, 2025-03-30:
the same

CostaliyA, 2025-03-30:
`export MUJOCO_GL="glx"` sloves my error. Thanks to  https://github.com/UT-Austin-RPL/Lotus/issues/4#issuecomment-2564411645 !

vineet2104, 2025-04-01:
What is your Ubuntu version? I have 22.04 and `export MUJOCO_GL="glx"` does not solve the issue for me. Was anybody able to solve this?

CostaliyA, 2025-04-02:
> What is your Ubuntu version? I have 22.04 and `export MUJOCO_GL="glx"` does not solve the issue for me. Was anybody able to solve this?

My Ubuntu version is 20.04.

CostaliyA, 2025-04-02:
> What is your Ubuntu version? I have 22.04 and `export MUJOCO_GL="glx"` does not solve the issue for me. Was anybody able to solve this?

 I guess this is a robosuite issue, you can find something in that repo.

zhong-hao-zhao, 2025-06-04:
If I want to perform inference without a graphical user interface, how should I set it up? I am doing inference on CentOS 7 but encountering the same issue.

Boltzmachine, 2025-08-31:
Same problem

huang-fuxian, 2025-12-17:
I  solved my problem, modify file"/miniconda3/envs/perlib312/lib/python3.12/site-packages/robosuite/renderers/context/egl_context.py"  , 44 line
from:
if selected_device is None:
to:
if selected_device is None or selected_device == "":