Lifelong-Robot-Learning/LIBERO #61: custom object

wangair, 2025-03-11:
Thank you for the contributions you have made! I have created the custom object and generated a bddl file.But I get Keyerror when I run collect_demonstration.py using this bddl file.What should I do about it？

zhuyifengzju, 2025-03-13:
Hi, can you describe how you are doing the custom object? For your reference, there is a [running example](https://github.com/Lifelong-Robot-Learning/LIBERO/blob/master/notebooks/custom_object_example.ipynb) you can refer to

wangair, 2025-03-17:
Thank you for your reply！I followed the notebooks/custom_object_example.ipynb，the pddl file is then generated.I used the pddl file as the collect_demonstration.py parameter, and met an error "KeyError: 'try_cup'".

lostgreen, 2025-04-12:
@wangair Hi, have you solved this problem? I met the same question.

eliotjones1, 2025-04-13:
I ran into this error as well. I can't say for certain if this is your error as well, but the `@register_object` decorator is quite picky. It expects the object class to be in CamelCase, and then turns it into snake_case. So for example, I had a difficult time with BluePlate, which I was referring to as "blueplate", but instead it should've been "blue_plate". If you share the relevant code blocks as well that might be helpful

lostgreen, 2025-04-21:
Yes, it's very strict. My error was in "states".
```
 @property
    def init_states(self):
        states = [
            ("On", "red_sticker_1", "kitchen_table_red_sticker_init_region"),
            ("On", "akita_black_bowl_1", "kitchen_table_akita_black_bowl_init_region"),
            ("On", "wooden_cabinet_1", "kitchen_table_wooden_cabinet_init_region"),
        ]
        return states
```
Only this format can be admitted. My first attempt uses ` ("On", "red_sticker", "kitchen_table_red_sticker_init_region")`, then it raise a `key error 'red sticker'`, it should be followed by it's index, like `red_sticker_1`