Lifelong-Robot-Learning/LIBERO #62: Regarding the Redefinition of Items

MstarLioning, 2025-03-16:
Hello! Thank you very much for the support your work has provided to my studies! However, I now have a question to ask: If I want to change the color of an object (for example, dye a bowl yellow), is there any method available to achieve this? Also, could you please let me know what format and which file I need to prepare if I want to define a new object? Thank you very much! I look forward to your reply!

MstarLioning, 2025-03-16:
To add, I have already found the corresponding objects in libero/libero/assets/stable_scanned_objects, but for instance, for akita_black_bowl, I see that there are two versions of the obj file, one labeled 'col' and the original version. Should I modify both? Thank you!

zhuyifengzju, 2025-03-17:
If you just want to modify visual appearance, you do not need to modify 'col', because it is for checking collision.

zhuyifengzju, 2025-03-17:
You can take a look at robosuite's wrapper on visual randomization to see if there are any functionality that fits your purpose.

MstarLioning, 2025-03-17:
Thanks！ I understand what you mean.I have figured out how to control the objects in the assets.

closed 2025-03-17 by MstarLioning