Lifelong-Robot-Learning/LIBERO #64: Where Can I Find the XML File Containing Environment Information in Robosuite?

ByeongGyu-Kim, 2025-03-18:
I have checked the available data, as shown in the attached image, and it seems to be the information used for fine-tuning. However, I am looking for the XML file that stores environment information in Robosuite, such as the position of objects like bowls, drawers, and other elements.

Where can I find this XML file that defines the environment configuration in Robosuite?

[image]

closed 2025-03-19 by ByeongGyu-Kim

pablovalle, 2025-04-10:
And where are the intial states saved? I think we need some initial states in order to initialize the environment, right?

kingdom-of-warriors, 2025-04-10:
That's right. Initial states are saved in "libero/libero/init_files/", and `init_states = task_suite.get_task_init_states(task_id)` can return this path.