Lifelong-Robot-Learning/LIBERO #70: The units of hdf5 data

nanamma, 2025-03-26:
I would like to know the units of the action values in the HDF5 file. For example, is the position data in meters (m) or centimeters (cm)? And is the rotation data in radians (rad) or degrees (°)?

Cranial-XIX, 2025-03-27:
It is meters, and rad

nanamma, 2025-03-27:
thanks

I would also like to ask about the physical meaning of each dimension in the data. When I simply accumulate the action, I get results in a strange range.
Is the current data relative or absolute? If it is relative, how can it be converted to absolute values?
If I wanna control with absolute positions or angles, how should I configure env and pass the action?

AniruthSuresh, 2025-08-12:
> thanks
>
> I would also like to ask about the physical meaning of each dimension in the data. When I simply accumulate the action, I get results in a strange range. Is the current data relative or [...]

Facing the same issue . Could you find a solution or a fix for the same ?

hdzwxd-prog, 2025-12-31:
same problem

when i plot {sum(action) + initial state} and the profile of "state" in the dataset i will get this picture

[image]

when i plot {sum(action)/100 + initial state} and the profile of "state" in the dataset i will get this picture
  

[image]

i don't think the unlinear characteristic of the dynamic cause the issue. I am convinced that robotic arms are capable of executing movements with sufficient precision. In practical applications, the positioning error of a typical robotic arm is at the sub-millimeter level. My suspicion is that the data has undergone a process similar to normalization.