Lifelong-Robot-Learning/LIBERO #76: Libero_90 "LIVING_ROOM_SCENE1_pick_up_the_tomato_sauce_and_put_it_in_the_basket_demo.hdf5" file truncated error

Zzfiv3, 2025-04-21:
Hi!
When I want to read the file "LIVING_ROOM_SCENE1_pick_up_the_tomato_sauce_and_put_it_in_the_basket_demo.hdf5" in Libero_90 datasets, the code is shown below:
```
print(f"Processing {hdf5_path}")
        try:
            hdf5_file = h5py.File(os.path.join(base_dir, hdf5_path), 'r', swmr=False, libver='latest')
            demos = list(hdf5_file["data"].keys())
        except Exception as e:
            print(f"Error processing {hdf5_path}: {e}")
            continue
```

I found the output has a truncated error: 
```
Processing LIVING_ROOM_SCENE1_pick_up_the_tomato_sauce_and_put_it_in_the_basket_demo.hdf5
Error processing LIVING_ROOM_SCENE1_pick_up_the_tomato_sauce_and_put_it_in_the_basket_demo.hdf5: Unable to synchronously open file (truncated file: eof = 590254080, sblock->base_addr = 0, stored_eof = 806284784)
```
Is anyone in the same situation as me, help!

Kai-Sheng-caesar, 2025-05-17:
Hi, I have the same problem, do you have any solution?

Kai-Sheng-caesar, 2025-05-20:
OK，thank you very much!