Lifelong-Robot-Learning/LIBERO #91: AttributeError: 'NoneType' object has no attribute 'eglQueryString'

xinyuezhang616, 2025-06-29:
when I run "from libero.libero.envs import OffScreenRenderEnv, SubprocVectorEnv, DummyVectorEnv" it has an error

[robosuite WARNING] No private macro file found! (macros.py:53)
[robosuite WARNING] It is recommended to use a private macro file (macros.py:54)
[robosuite WARNING] To setup, run: python /home/zhangxinyue/.conda/envs/ript/lib/python3.10/site-packages/robosuite/scripts/setup_macros.py (macros.py:55)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/zhangxinyue/LIBERO/libero/libero/envs/__init__.py", line 1, in <module>
    from .bddl_base_domain import TASK_MAPPING
  File "/home/zhangxinyue/LIBERO/libero/libero/envs/bddl_base_domain.py", line 3, in <module>
    import robosuite.utils.transform_utils as T
  File "/home/zhangxinyue/.conda/envs/ript/lib/python3.10/site-packages/robosuite/__init__.py", line 1, in <module>
    from robosuite.environments.base import make
  File "/home/zhangxinyue/.conda/envs/ript/lib/python3.10/site-packages/robosuite/environments/__init__.py", line 1, in <module>
    from .base import REGISTERED_ENVS, MujocoEnv
  File "/home/zhangxinyue/.conda/envs/ript/lib/python3.10/site-packages/robosuite/environments/base.py", line 12, in <module>
    from robosuite.utils.binding_utils import MjRenderContextOffscreen, MjSim
  File "/home/zhangxinyue/.conda/envs/ript/lib/python3.10/site-packages/robosuite/utils/binding_utils.py", line 58, in <module>
    from robosuite.renderers.context.egl_context import EGLGLContext as GLContext
  File "/home/zhangxinyue/.conda/envs/ript/lib/python3.10/site-packages/robosuite/renderers/context/egl_context.py", line 32, in <module>
    from mujoco.egl import egl_ext as EGL
  File "/home/zhangxinyue/.conda/envs/ript/lib/python3.10/site-packages/mujoco/egl/__init__.py", line 31, in <module>
    from mujoco.egl import egl_ext as EGL
  File "/home/zhangxinyue/.conda/envs/ript/lib/python3.10/site-packages/mujoco/egl/egl_ext.py", line 27, in <module>
    from OpenGL import EGL
  File "/home/zhangxinyue/.conda/envs/ript/lib/python3.10/site-packages/OpenGL/EGL/__init__.py", line 2, in <module>
    from OpenGL.raw.EGL._types import *
  File "/home/zhangxinyue/.conda/envs/ript/lib/python3.10/site-packages/OpenGL/raw/EGL/_types.py", line 87, in <module>
    raw_eglQueryString = _p.PLATFORM.EGL.eglQueryString
AttributeError: 'NoneType' object has no attribute 'eglQueryString'

How can I solve it?
I have run "pip install -e ." in my conda env.

Tonghe-Zhang, 2025-06-29:
Maybe try this if you have sudo priviledge. 
```bash
sudo apt-get install libegl-dev
python <PATH_TO_YOUR_CONDA>/envs/<YOUR_ENV_NAME>/lib/python3.8/site-packages/robosuite/scripts/setup_macros.py
```
I met similar issue when robomimic environment is not properly installed. Hope it helps.

Yz-carrot, 2025-10-29:
Follow GPT5, I solved it by steps below:
1. install something:
`apt-get update && apt-get install -y \
  libglvnd0 libgl1 libegl1 libgles2 libdrm2 libgbm1`

2. set the environment:
`export MUJOCO_GL=egl`